oliverlee/phobos #128: Clearer project structure

mickvangelderen, 2017-03-07:
The structure of the project is decent but could be better. I have trouble figuring where all the source files come from for an executable/library.

1. A clear separation between executables that need to run on the microcontroller, and executables that run on the host pc.
2. The `external/` folder is essentially a collection of common code organized in libraries. We could also put our own code in such libraries (without creating separate git repositories for them, but allowing the possibility). 
3. Limit nesting of directories. We can put applications and demos in the same directory and simply post-fix the demos with `_demo`.

```
dependencies/
    chibios/ (git module)
    phobos_chibios_common/
    cobs/
microcontroller/
    simulator/
    usb_serial_demo/
desktop/
    log_simulator/
    plot_x/
    plot_y/
```

Perhaps we should use two top-level `CMakeLists.txt` files instead of one.

oliverlee, 2017-03-07:
RFC = Refactor code?

This is a good idea, particularly for microcontroller vs. desktop code, as is two top-level CMakeLists.txt files.

mickvangelderen, 2017-03-07:
RFC = Request For Comments, but after I changed the title I used labels so RFC can be removed again.

oliverlee, 2017-06-27:
Let's revisit this on 5 July.

closed 2020-07-20 by mickvangelderen