oliverlee/phobos #143: Crash when moving average sum overflows

mickvangelderen, 2017-03-09:
If the sum wraps then depending on what is being filtered, people might get injured.

https://github.com/oliverlee/phobos/blob/master/src/filter/movingaverage.hh#L28

oliverlee, 2017-03-09:
Not sure how you can better handle it. It adds the difference of newest and oldest to the running sum.

oliverlee, 2017-03-09:
This is currently used with floats and not at risk of crashing with current code. Changing the milestone to a later date.

mickvangelderen, 2017-03-09:
Safer implementation: Store the difference with the expected value (delta hereafter) instead of absolute values. The sum of the deltas will then hover around 0, assuming the expected value doesn't change. Update the expected value (and thus all deltas) only when `abs(expected_value) > MAX_VALUE_OF_TYPE / N` or some other criteria.

oliverlee, 2017-05-17:
We could also potentially use a larger data type for `m_sum`.

I'll look at where this filter is used as it may not be an issue. Some of the analog signals are 12-bit so I can't imagine ever overflowing a 16-bit integer with the data buffer sizes we use.

mickvangelderen, 2017-05-18:
A warning should be added to the documentation of this moving average filter that it does not protect against the sum overflowing. So if it is possible that the sum of the values in the filter exceeds the maximum value that the T type can store, a different implementation should be used.

commit 122a2ab4 (2017-06-28): Merge pull request #236 from oliverlee/document-moving-average-overflow

Document moving average overflow

Fixes #143
closed 2017-06-28 by oliverlee