oliverlee/phobos #147: Increase motor actuation rate

oliverlee, 2017-03-10:
We should running the simulation/actuation loop at 1 kHz or at least 500 Hz for haptics.

oliverlee, 2017-03-10:
Recent protobuf streams show the timing to be:
```
timing {
  computation: 63904
  transmission: 32899
}
```
These values are calculated with the ChibiOS TM module which uses the realtime counter. At 168 MHz, the resulting times are:
computation: 0.372 ms
transmission: 0.208 ms
so changing the loop rate to 1 kHz should be 👍

mickvangelderen, 2017-03-14:
I don't get it. Is each iteration triggered by a timer? Do you increase the processor clock frequency or the timer frequency?

oliverlee, 2017-03-14:
Each iteration of the simulation loop is triggered by a virtual timer: https://github.com/oliverlee/phobos/blob/f2c686b29f62306f04b3828090c80b70a71f1ddf/projects/flimnap/main.cc#L331-L337

The frequency of the virtual timer is controlled by `CHIBIOS_CFG_ST_FREQUENCY` in CMake but PR #154 wants to revert this and change configuration back to `CH_CFG_ST_FREQUENCY` in `chconf.h`.

PR #152 increases simulation loop rate by changing the thread sleep argument. However, this results in poor loop timing precision as the minimum system time delta is 2 system ticks (the minimum number of ticks that is safe to specify in a timeout directive), so the system tick frequency is also increased. 

The two fields in the timing submessage are calculated using the ChibiOS time measurement API: http://chibios.sourceforge.net/docs3/rt/group__time__measurement.html
This uses the realtime counter which runs at 168 MHz (stated here: https://www.olimex.com/Products/ARM/ST/STM32-H405/ and also in this file: https://github.com/oliverlee/ChibiOS/blob/9b5a700b8382bef5807c80fe4cbd4d2a1f3cf5ef/os/hal/ports/STM32/STM32F4xx/hal_lld.h)
You can use the preprocessor define STM32_SYSCLK to get the clock speed (but it's 168000000).

These values (roughly) show we are not exceeding the total time we budget for each iteration (1 ms for 1 kHz, 2 ms for 500 Hz).

PR #152 (2017-03-14): Increase simulation loop rate
Fixes #147 
- [x] merge #150 
- [x] merge #149 
- [x] test on hardware and confirm no deadline misses: looks good enough to me!
closed 2017-03-14 by oliverlee