oliverlee/phobos #148: Fix ADC events

oliverlee, 2017-03-10:


mickvangelderen, 2017-03-14:
What are ADC events?

oliverlee, 2017-03-14:
ADC is the analog to digital converter. The Analog class is supposed to broadcast two events: `adc_eventflag_complete` and `adc_eventflag_error`. The events are defined here: https://github.com/oliverlee/phobos/blob/f2c686b29f62306f04b3828090c80b70a71f1ddf/src/analog.cc#L29-L30

An example using events is here: https://github.com/oliverlee/phobos/blob/f2c686b29f62306f04b3828090c80b70a71f1ddf/demos/adc/main.cc

oliverlee, 2017-05-17:
This should probably be fixed for the steer-by-wire bike I think as this bug isn't causing problems for the static simulator at the moment.

oliverlee, 2017-07-06:
This doesn't appear to be necessary at the moment.

closed 2017-07-06 by oliverlee