oliverlee/phobos #162: Update toolchain download location

oliverlee, 2017-03-16:
As previously announced all new binary and source packages will not be released on Launchpad henceforth, they can be found on:
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm.

oliverlee, 2017-03-16:
Updating the toolchain will bring us to gcc-6 which has cool color output when you compile:

[image]

oliverlee, 2017-03-16:
Note that std::clamp still does not exist.

Reinterpret casts are not allowed in constexpr so some redefinitions will be necessary. For example,
```cpp
constexpr EncoderConfig RLS_ROLIN_ENC_INDEX_CFG = {
    .z = PAL_LINE(GPIOA, GPIOA_PIN2),
    .counts_per_rev = 152000,
    .filter = EncoderConfig::filter_t::CAPTURE_64, // 64 / 42 MHz (TIM5 on APB1) = 1.52 us for valid edge
    .z_count = 18407
};
```
will need to change to
```cpp
constexpr EncoderConfig RLS_ROLIN_ENC_INDEX_CFG = {
    .z = PAL_LINE(GPIOA_BASE, GPIOA_PIN2),
    .counts_per_rev = 152000,
    .filter = EncoderConfig::filter_t::CAPTURE_64, // 64 / 42 MHz (TIM5 on APB1) = 1.52 us for valid edge
    .z_count = 18407
};
```
as ChibiOS defines `GPIOA` in `os/hal/ports/STM32/LLD/GPIOv2/pal_lld.h` using a reinterpret cast to convert an unsigned int to a pointer:
```c
#define GPIOA                           ((stm32_gpio_t *)GPIOA_BASE)
```
where type of `GPIOA_BASE` is `uint32_t`.

oliverlee, 2017-03-16:
We could also change the standard language to `c++1z`

oliverlee, 2017-06-15:
The compiler also learns the option `-Wno-lto-type-mismatch` which we can use to disable all the annoying protobuf warnings we get when LTO is enabled.

mickvangelderen, 2017-06-20:
What are the implications for developers on this project? Is downloading the new version of the toolchain all that is required for each development machine?

We musn't forget to update the readme.

oliverlee, 2017-06-20:
- [ ] update toolchain on dev machines
- [ ] update toolchain on travis
- [ ] update readme
Can't think of anything else.

oliverlee, 2017-06-20:
I also think is low priority so we can do it sometime after the July demo.