oliverlee/phobos #168: Merge log visualisation prototype into bikesim

mickvangelderen, 2017-04-18:
The work from #167 needs to be merged into bikesim. We'll have to see how this goes.

mickvangelderen, 2017-04-18:
The original "bikesim" project is a bit of a mess. There is a lot of unused and duplicated code, stale comments. It depends on some very low quality assets. The unity project configuration is somehow messed up because some scripts won't compile because of duplicated implementations of the standard ConcurrentQueue class in different assemblies. Due to these problems, merging the useful parts of bikesim into a new project turned out to be easier than updating the bikesim project itself. 

I copied over the bicycle game object and the code that updates its transform. This part needs to be tested using the physical bike simulator.

oliverlee, 2017-04-19:
The project didn't compile with the newest version of Unity? I haven't run into that issue in the past.

mickvangelderen, 2017-04-20:
> The project didn't compile with the newest version of Unity? I haven't run into that issue in the past.

It doesn't when you try to use `ConcurrentQueue<T>`.

oliverlee, 2017-04-25:
I'm assuming this got fixed since you were able to run the visualization today?

mickvangelderen, 2017-04-25:
The new visualisation project is hosted here https://gitlab.com/bikelab/phobos-visualiser

closed 2017-04-25 by mickvangelderen