oliverlee/phobos #169: Bicycle simulation stops when velocity is zero

oliverlee, 2017-04-25:
If we are running the simulation and stop pedaling, the velocity of the bicycle goes to zero. This also stops the state of the bicycle from changing as seen here:

```cpp
    if (m_dt == static_cast<real_t>(0)) { // discrete time state does not change
        m_Ad.setIdentity();
        m_Bd.setZero();
    } else {
```
https://github.com/oliverlee/bicycle/blob/0ffe6f2ba4de25b1c01fb78ef64168aaae1deadf/src/bicycle/bicycle.cc#L276-L279

We should allow the simulation to advance in time for zero velocity and determine how to handle special cases (e.g. start of the simulation before the rider starts pedaling).

oliverlee, 2017-04-25:
The above code stops the simulation from running if `dt = 0`, not `v = 0`. In fact, it is more likely that we limit the state when simulating the bicycle here:
```cpp
    { // limit allowed bicycle state
        state_t x = m_observer.state();

        auto limit_state_element = [&x](model::Bicycle::state_index_t index, real_t limit) {
            const real_t value = model::Bicycle::get_state_element(x, index);
            if (std::abs(value) > limit) {
                model::Bicycle::set_state_element(x, index, std::copysign(limit, value));
            }
        };

        real_t roll_angle = model_t::get_state_element(m_observer.state(),
                model_t::state_index_t::roll_angle);
        if (std::abs(roll_angle) > constants::pi) {
            // state normalization limits angles to the range [-2*pi, 2*pi]
            roll_angle += std::copysign(constants::two_pi, -1*roll_angle);
        }

        limit_state_element(model_t::state_index_t::roll_rate, roll_rate_limit);
        limit_state_element(model_t::state_index_t::steer_rate, steer_rate_limit);

        m_observer.set_state(x);
    }
```
https://github.com/oliverlee/phobos/blob/99f9db1afbec528f7189ef97663b91626f360698/projects/src/simbicycle.hh#L93-L114

oliverlee, 2017-04-25:
@mickvangelderen Can you check if the bike starts falling over with `v=0`? I think it should fall until the roll angle limit is reached, then you would see the state freeze.

oliverlee, 2017-05-17:
This will also need to be handled for the steer-by-wire bicycle which may not have the same solution.

mickvangelderen, 2017-05-18:
[image]

mickvangelderen, 2017-06-27:
The LQR controller does a good job at solving this issue. See #224.

oliverlee, 2017-06-27:
#224 Hasn't been merged it. I'd like to leave this open until it has been merged (hopefully later this afternoon).

PR #224 (2017-06-27): Add low speed roll assistance
This PR reimplements https://github.com/oliverlee/phobos/pull/191

Fixes #169, https://github.com/oliverlee/phobos/issues/218

~~This PR depends on #213~~
commit 9f000b5d (2017-06-27): Merge pull request #224 from oliverlee/add-low-speed-assistance

Add low speed roll assistance

Fixes #169,  fixes #218
closed 2017-06-27 by oliverlee