oliverlee/phobos #174: Improve visuals

oliverlee, 2017-05-04:
This needs to be better. I guess we should add some trees and a road.

mickvangelderen, 2017-05-11:
@oliverlee Any particular path for the road?

oliverlee, 2017-05-11:
If you have time to work on this, ask Arend. I'm not sure what he wants as I think whatever is implemented is already sufficient.

oliverlee, 2017-05-15:
The general topology should look like a 3x3 grid:

    +---------+---------+------------+
    |         |         |            |
    |         |         |            |
    +---------+---------+------------+
    |         |         |            |
    |         |         |            |
    |         |         |            |
    +---------+---------+------------+
    |         |         |            |
    |         |         |            |
    |         |         |            |
    |         |         |            |
    +---------+---------+------------+
- The grid should _not_ be evenly spaced.
- The perimeter of the grid should be 1 km.
- The corners of the grid should be curved (radius of curvature is not strictly defined right now)
- The road should be composed of a two lane bike path: https://en.wikipedia.org/wiki/File:Bicycle_lane_Dronten.JPG

Trees can be populated throughout so that it "looks good". Ground that is not road can be grass or a grass texture.

oliverlee, 2017-05-30:
Maybe we could take a look at this as well: https://github.com/fcl-engaging-mobility/UnityScripts

mickvangelderen, 2017-06-23:
[image]
First iteration

closed 2017-06-23 by mickvangelderen

oliverlee, 2017-06-24:
Implemented here: https://gitlab.com/bikelab/phobos-visualizer/commit/4c315acc6d99940e6e612ee1f0f77ca1739562b8