oliverlee/phobos #175: Display fps on screen

oliverlee, 2017-05-04:
Why not? People will ask about this (as well as the time it takes to transmit a message and draw it to screen but that's a bit trickier to calculate).

mickvangelderen, 2017-05-11:
Interesting frequencies include:

- Drawing frequency (Update)
- Update frequency (Fixed Update)
- Received message frequency

oliverlee, 2017-05-11:
Is anything done (or will ever be done) in fixed update?

Definitely update, after all this gives us fps.
Possibly received message frequency, but maybe as X messages/second as we would be receiving simulation state messages at 1 kHz and pose messages at 120 Hz.

mickvangelderen, 2017-05-12:
We should read from the serial port in fixed update. Getting all the messages is critical for logging. If we read from serial in update and the frame rate drops, flimnap will be dropping messages.

oliverlee, 2017-05-15:
I figured a low priority background thread would be continuously reading from the serial port, not necessarily at any fixed rate.

mickvangelderen, 2017-05-16:
That makes more sense, yes.

closed 2017-06-23 by mickvangelderen