oliverlee/phobos #176: Create binary allowing device name input

oliverlee, 2017-05-04:
Not sure if there should be a splash screen? What about if the microcontroller is reset and the device name changes? Ideally we wouldn't need to restart the visual environment and could just change the device name to read from.

mickvangelderen, 2017-05-11:
Ties into https://github.com/oliverlee/phobos/issues/177

mickvangelderen, 2017-05-16:
This thing works.
[image]
It is horrible from a user standpoint however.

- no feedback
- no suggestions/autocomplete
- no navigation with the TAB key
- no autofocus
- no recent ports

It does have a [platform specific default port](https://gitlab.com/bikelab/phobos-visualizer/blob/b56d249211a3188e9ed0c9cab92b6b611b101f71/Assets/Scripts/Configuration.cs#L16-24)

mickvangelderen, 2017-05-16:
Not restarting the visualizer will be a challenge https://github.com/oliverlee/phobos/issues/185

mickvangelderen, 2017-05-18:
Design for the screen:

```
Serial Device Port: 
⊙ Detected: [ /dev/ttyACM0 | V ]
○ Manual:   _________________

[ Connect ]
```

We will look for files matching a regex (for example `\/dev\/ttyACM\d+`), specific to the platform we are running on. On windows we could perhaps use `SerialPort.GetPortNames()` because the serial devices are probably not files on the filesystem.

The detection should run periodically while the screen is active.

When 1) no data is coming from the serial port for more than 100ms, 2) we cannot succesfully interpret the incoming data, or 3) the user presses the disconnect button, we should close the connection and all associated resources and reopen the port selection screen.

mickvangelderen, 2017-05-18:
Took a while but after some frustration, forum hunting and tutorial watching I figured out how auto layout (somtimes) works in Unity. Feel failrly confident I can add other configuration options, buttons and what not in the future with relative ease.

[image]

mickvangelderen, 2017-06-21:
Works really well right now. 
[image]

closed 2017-06-21 by mickvangelderen