oliverlee/phobos #177: Restart simulation on button press

oliverlee, 2017-05-04:
Restart bicycle simulation on button press. Using the reset button may change the serial device name.

mickvangelderen, 2017-05-11:
The restarting already works. The visualizer needs to recognize that the serial connection has been closed. When that happens we can either:

- Autodetect open ports depending on the platform and either present them to the user or select one automatically
- Query the new port from the user

oliverlee, 2017-05-11:
The less the user has to do the better so prefer autodetect and autoselect.

The only issue I see if handling multiple serial devices (i.e. ones that are plugged in but are not sending related to phobos). We could connect to a device and wait for a short time for valid data. If none is received, try the next device. If we try all available devices and no data is received, ask the user to check that the device is actually connected.

oliverlee, 2017-05-15:
> The restarting already works. The visualizer needs to recognize that the serial connection has been closed. 

We want to be able to restart by pressing the round push button and not the square reset button. When the reset button is pressed, the GPIO lines float so the actuator behavior is undefined and any configuration data is lost (e.g. absolute position of the steer encoder).

Pushing the push button can just reinitialize the state of the simulation and possibly flush any messages in the transmit queue.

oliverlee, 2017-05-15:
I just realized the steer encoder uses PA0, PA1 for A, B channels which overlaps with the push button which is also on PA0.

Because of this, we can:
 - Not have push button functionality and rely on board reset.
 - Move steer encoder from TIM5 to TIM4. Although TIM4 is 16-bit timer (only TIM2 and TIM5 are 32-bit), we use less than half the full range of motion (I think) so this should still work.

mickvangelderen, 2017-05-16:
I have some, perhaps silly, questions. 

1. What are PA0 and PA1.
2. What are the A and B channels?
3. What does a timer do for a steer encoder?
4. Why would we want to use TIM4 instead of TIM5?

oliverlee, 2017-05-16:
> I have some, perhaps silly, questions.

not silly

> What are PA0 and PA1.

GPIO (general purpose input output) Pad A Pin 0 and GPIO Pad A Pin 1
GPIO are digital input and output you can control. They are clustered in groups, or pads. On some board layouts, the pins in the the same pad will physically grouped together.

> What are the A and B channels?

A digital encoder transmits two square waves in order to track velocity and position. A single square wave would only allow tracking of velocity and position if the direction never changes. In order to determine direction, the signals are 90 degrees out of phase. These two signals are commonly denoted A channel and B channel.

> What does a timer do for a steer encoder?

The encoder sends square waves which we need to read in order to determine the position. There are 3 general ways to keep track of the encoder counts in software:
 1. Use GPIO to poll the two signals. The polling rate needs to be much faster than the speed of the encoder signal.
2. Use interrupts triggering on signal edges to change the encoder count. This can adversely affect system timing depending on how often edges are encountered and interrupt latency.
3. Use a hardware timer in encoder mode. The external signals are essentially wired to a clock. This is the most efficient method.

> Why would we want to use TIM4 instead of TIM5?

In a microcontroller, the GPIO is multiplexed with other functions. For our case, GPIOA_PIN0 can also be used as TIM5_CH1 (channel 1 of timer 5). Since GPIOA_PIN0 is wired to the push button, we can only use one or the other. We can connect the A, B channels of the steer encoder to TIM4_CH1 and TIM4_CH2 through pins GPIOB_PIN6 and GPIOB_PIN7, allowing use of GPIOA_PIN0 for the push button instead of a channel of TIM5.

Only TIM2 and TIM5 are 32-bit timers in our architecture, all other timers are 16-bit. As the steer encoder resolution is 152000 counts per revolution, normally we should tie it to a 32-bit timer. However, I don't believe we actually exceed a 16-bit range as the handlebars cannot be spun a full revolution; I think it's roughly 90 degrees full range before the mechanical stops.

mickvangelderen, 2017-05-16:
Thanks :smile:.

On the topic of restarting the simulation on button press. It seems to me that using the push button as an intermediate solution is not as little development effort as we initially thought. Perhaps the time is better spent on two way communication?

oliverlee, 2017-05-17:
Yeah, it sounds better to _not_ spend time hacking a temporary solution (i.e. physical button). I think we should still have Unity autodetect+autoconnect to the new serial port because I imagine that can't be more than a day's work. However, the motor behavior on reset is something we'll just have to live with until two way communication is implemented.

mickvangelderen, 2017-05-18:
> I think we should still have Unity autodetect+autoconnect to the new serial port

Yes this should be implemented, but it is not as simple as I had hoped due to the fact we are using a virtual serial connection over USB. Also we have to implement it for windows as well because that is where I expect most people to want to run the visualizer on eventually. It is more than a day's work.

oliverlee, 2017-06-14:
Port autodetect is implemented https://gitlab.com/bikelab/phobos-visualizer/commits/fancy-menu
Autoconnect will not be implemented as the user must manually select the port.

closed 2017-06-21 by oliverlee