oliverlee/phobos #184: Create phobos-visualizer build script

mickvangelderen, 2017-05-12:
We need a script that builds binaries for OS X, Windows and Linux using Unity and compresses the applications so we can attach them to phobos releases. 

#171

oliverlee, 2017-05-15:
Looks like setting up a unity build is easy enough:
http://blog.stablekernel.com/continuous-integration-for-unity-5-using-travisci
https://jonathan.porta.codes/2015/04/17/automatically-build-your-unity3d-project-in-the-cloud-using-travisci-for-free/

I don't know anything about copying the built binaries to somewhere. Would we create a nightly release and attach the binaries to that? Do you have experience with that?

mickvangelderen, 2017-05-16:
Thought about it, never written a working solution though.

oliverlee, 2017-05-16:
Looks like the Github allows something for releases (not branches): https://docs.travis-ci.com/user/deployment/releases/

mickvangelderen, 2017-07-25:
Uploading attachments to releases on gitlab: https://gitlab.com/gitlab-org/gitlab-ce/issues/18486

Example scripts: http://answers.unity3d.com/questions/9382/build-from-script.html

mickvangelderen, 2017-08-11:
Been working on this but it is very hard to activate a unity install from the command line. Activation is required for any command including building. The reason for this is probably that Unity has their own "Cloud building" solution. 

Putting more time into this is not worth it. A build script is still useful.

mickvangelderen, 2017-08-11:
Fixed by https://gitlab.com/bikelab/phobos-visualizer/commit/92a93d971a8c54fea6225ce8f7c425389c4d17e2

mickvangelderen, 2017-08-11:
Updated release v0.2 with win|lin|mac 32|64 builds.

closed 2017-08-11 by mickvangelderen