oliverlee/phobos #185: Make phoviz plug and play

mickvangelderen, 2017-05-16:
If the usb cable is disconnected, the visualizer does not stop running. This might be due to the way it has been programmed. Not sure what to do about this.

http://stackoverflow.com/questions/9835881/releasing-a-unplugged-virtual-serial-port

oliverlee, 2017-05-17:
If the usb cable is disconnected, we'll stop receiving messages. Even though the visualizer is still running, the bicycle pose isn't getting updated, which we can easily see (also timestamp if that's being displayed). I don't really see that as a problem?

I haven't thought much about the desired behavior if the usb cable is disconnected. What are your thoughts?

oliverlee, 2017-05-17:
Although I guess I'm not clear on when logging messages to file should stop and start.

oliverlee, 2017-05-17:
After seeing https://github.com/oliverlee/phobos/issues/176, maybe we should discuss phoviz behavior over a video chat.

mickvangelderen, 2017-05-18:
We'll take the "disconnect when no messages have been received for x seconds" approach.

mickvangelderen, 2017-06-27:
If the simulation keeps running and the visualizer is stopped, the visualizer is able to catch up.

If the visualizer keeps running and the simulation is stopped or if the cable is simply disconnected, the visualizer keeps displaying the last state it received. The status panel indicates no messages are being received.

What happens when the cable is re-attached needs to be tested. The simulation should just continue after https://github.com/oliverlee/phobos/pull/197. Hopefully, the visualization will simply continue but it depens on the C# SerialPort class.

mickvangelderen, 2017-06-27:
When disconnecting and reconnecting the USB cable while the simulation and visualization are running, the simulation seems to continue (rapid blinking green led) but the visualizer does not start processing messages again.

Re-starting the visualizer works.

mickvangelderen, 2017-06-27:
Restarting the visualizer to handle USB disconnects (which should not occur under normal operating conditions) is a fine solution.

closed 2017-06-27 by mickvangelderen