oliverlee/phobos #186: Determine logging format

oliverlee, 2017-05-17:
So far, I assume logs will be the COBS framed serialized protobuf data. I'm not sure this is the best approach storing the data, as we'll need to do a bunch of processing to actually look at it with other tools. And this will be a problem if anyone else besides us (@oliverlee, @mickvangelderen) is working with the data at a later date, particularly since it encapsulates the framing protocol and requires the protobuf schema.

It seems like hdf5 seems to be a hip format to store stuff. We could write a tool that transforms the data from framed serialized protobuf to hfd5 (and possibly do this directly in Unity).

oliverlee, 2017-05-17:
As replay requires a protobuf stream, I don't think we should do this in Unity.

Issue https://github.com/oliverlee/phobos/issues/172 relates to actually logging in Unity.

oliverlee, 2017-05-17:
This was an interesting read: http://cyrille.rossant.net/should-you-use-hdf5/

mickvangelderen, 2017-05-18:
I don't know about any well known binary arbitrary data file formats. We can support multiple formats once we have the conversion tools easily so experiment away :smile:.

oliverlee, 2017-05-30:
We will have two types of files:
- logging format: framed serialized protobuf, file extension of __.pbf__
- storage format: hdf5, file extension of __.h5__

mickvangelderen, 2017-06-01:
> logging format: framed serialized protobuf, file extension of .pbf

I'd like to see `.pb.cobs`, similar to `.tar.gz`.

oliverlee, 2017-06-14:
We need to go through the repo and rename files to __.pb.cobs__

oliverlee, 2017-07-06:
What we have works for now so maybe we'll reopen this if necessary.

closed 2017-07-06 by oliverlee