oliverlee/phobos #187: Determine latency for visuals

oliverlee, 2017-05-18:
We need at least a rough estimate on how long it takes Unity to draw the environment to screen (monitor or headset) as well as a latency for the sensor data to propagate to Unity/screen.

mickvangelderen, 2017-06-23:
@oliverlee is this still relevant?

oliverlee, 2017-06-24:
Yes.

Time for motion to be sampled, fed into Kalman filter + dynamics simulation, and a completed transmission from microcontroller to desktop should be < 1 ms from previous testing with a blocking `usbTransmit()` call in the simulation loop. Here we don't explicitly take into account the 'sampling time' of the encoder as we are using a timer in encoder mode, and pose messages aren't sent at 1 kHz, but I think a 1 ms upper bound for all the microcontroller stuff is reasonable.

Now we need to estimate the time from being read in Unity to photon output. This should be less than 13 ms as we consistently display 75 FPS, otherwise we would be dropping frames.

Looks like there are some tools to use to test out motion-to-photon latency:
https://www.reddit.com/r/oculus/comments/43p3a6/does_anybody_know_the_motiontophoton_latency_of/
https://forums.oculus.com/developer/discussion/43597/whats-your-motion-to-photon-latency
https://www.reddit.com/r/oculus/comments/55avya/high_motiontophoton_latency_how_to_find_the_cause/

oliverlee, 2017-06-24:
Also it looks like we should just buy CV1 to reach 90 FPS.

oliverlee, 2017-06-26:
Using the following commits:
1ac3acf154fb4a79309a687d3840f06c67babf7b
https://gitlab.com/bikelab/phobos-visualizer/commit/4c315acc6d99940e6e612ee1f0f77ca1739562b8

I observe an app motion-to-photon latency of ~ 23 ms.
Log file here: https://docs.google.com/spreadsheets/d/1be0ZcxpZIkWNshU5prqRswj9LfAd-1It74HtP4rDpAc/edit?usp=sharing

So roughly 24 ms total latency.

closed 2017-06-26 by oliverlee