oliverlee/phobos #193: wheel angle in viz too large

oliverlee, 2017-06-07:
Wheel value grows too large as isn't getting normalized at all.

mickvangelderen, 2017-06-12:
How is velocity better? In general, you can estimate the velocity from a few position samples but you need all velocity samples to find the position.

mickvangelderen, 2017-06-12:
I'm curious what happened that brought up this issue.

oliverlee, 2017-06-12:
The angular position of the rear wheel doesn't really mean much and is hard to make sense of it when shown on screen as a number, although it does help in rendering the wheels.

Displaying the forward velocity of the bicycle would be better as this a common metric people use when riding.

The value is shown on screen and grows away from zero when pedaling.

mickvangelderen, 2017-06-13:
Ah you are talking about the status panel. I see what you mean now.

PR #194 (2017-06-15): Normalize pose rear wheel angle
Take modulo of pose rear wheel angle with modulus two pi to prevent rear
wheel angle from continuously increasing.

Fixes #193
closed 2017-06-15 by oliverlee