oliverlee/phobos #207: Properly translate simulation message coordinate system to unity

mickvangelderen, 2017-06-20:
The simulation messages use the coordinate system described in Meijaard, Jaap P., et al. "Linearized dynamics equations for the balance and steer of a bicycle: a benchmark and review.". Unity uses a different coordinate system.

oliverlee, 2017-06-20:
Meijaard uses a right hand coordinate system. Unity uses a left hand coordinate system. I believe all you need to do is reverse one of the axis to correctly transform them. Take a look here, specifically lines 185-187: https://github.com/oliverlee/bikesim/blob/c7f15035ad2ac4123aca5a5e9cf07d6995176a43/bikesim/Assets/Scripts/BicycleController.cs#L177-L232

mickvangelderen, 2017-06-20:
Fixed with https://gitlab.com/bikelab/phobos-visualizer/commit/625e776f521bc69c978f44a7c4a1434a62c66f5e

closed 2017-06-20 by mickvangelderen