oliverlee/phobos #222: Motor jerkiness

oliverlee, 2017-06-22:
See here:
[image]
[image]

Data is from here:  https://github.com/oliverlee/phobos/pull/198#issuecomment-310365293

oliverlee, 2017-06-22:
This is fixed by setting the analog input low pass filter to 100 Hz in the kollmorgen workbench.

closed 2017-06-22 by oliverlee

mickvangelderen, 2017-06-23:
I have some questions to improve my understanding of the system:

1. Is the commanded steer velocity limited to +-100 deg/s?
2. Is it coincidence that the low pass filter is set to Hz and the velocity limit is 100 deg/s?
3. How did you determine 100 Hz is a good maximum frequency?

oliverlee, 2017-06-23:
> Is the commanded steer velocity limited to +-100 deg/s?

Yes this is set in the motor drive configuration. The drive accepts a velocity input from ±12V. We map this to ±100 deg/s.

> Is it coincidence that the low pass filter is set to Hz and the velocity limit is 100 deg/s?

Yes. The low pass filter cutoff frequency is not related to the velocity limit.

> How did you determine 100 Hz is a good maximum frequency?
Intuition. We don't expect the bicycle dynamics to be that high. The cutoff frequency was reduced to 100 Hz from 500 Hz.