oliverlee/phobos #228: Unable to build phoviz on windows

oliverlee, 2017-06-26:
Build fails due to compiler error:
```
The type or namespace name 'Concurrent' does not exist in the namespace 'System.Collections'. Are you missing an assembly reference?
```

mickvangelderen, 2017-06-27:
I implemented a concurrent bounded queue and am using it reliably in the `Live` scene. The `Replay` scene is still using some old code. This can be fixed relatively easily. 

The reason I haven't updated the `Replay` scene is because the COBS and Protobuf decoding for a pull based stream with timeouts of which we cannot use the underlying `Stream` class (serial port) and a pull based stream (file stream) is not easily abstracted.

mickvangelderen, 2017-06-27:
Moved issue to Improved Demo because the replay functionality is not essential.

mickvangelderen, 2017-06-27:
https://gitlab.com/bikelab/phobos-visualizer/commit/2d85b200915c19c2b20607c71157d45140151fc9

closed 2017-06-27 by mickvangelderen