oliverlee/phobos #229: Give rider roll angle feedback

mickvangelderen, 2017-06-27:
Roll of the entire camera mount is always disabled now. The camera mount object is following the bicycle object. 

One idea is to add a swivel object that keeps the camera flat while still having it move with the roll of the bike. 
[image]

Another possibility is a HUD element indicating roll.

mickvangelderen, 2017-06-27:
Other suggestions:

- Have the sky box *not* rotate with the world.
- Scale the roll rate down emperically.
- View the player actor (bicycle) in third person, even though that might cause other mismatches.

oliverlee, 2017-06-27:
I think we want to implement all 3, but not necessarily by Friday.

mickvangelderen, 2017-06-29:
Apart from the skybox everything is implemented and seems sufficient to provide feedback https://github.com/oliverlee/phobos/issues/233.

closed 2017-06-29 by mickvangelderen