oliverlee/phobos #239: Compress .pb.cobs logs

mickvangelderen, 2017-06-28:


oliverlee, 2017-06-29:
This isn't working
```
(dev) oliver@canopus:~/repos/phobos/scripts$ ./plot_sim.py ~/Downloads/2017-06-29T12h34m30sZ.pb.cobs
Traceback (most recent call last):
  File "./plot_sim.py", line 109, in <module>
    messages = load_messages(sys.argv[1])
  File "/Users/oliver/repos/phobos/scripts/load_sim.py", line 44, in load_messages
    return load.cobs_framed_log(filename, deserialize_callback, True)
  File "/Users/oliver/repos/phobos/scripts/phobos/load.py", line 67, in cobs_framed_log
    packets)
  File "/Users/oliver/repos/phobos/scripts/phobos/load.py", line 102, in __decode_multipacket
    message = callback(data)
  File "/Users/oliver/repos/phobos/scripts/load_sim.py", line 42, in deserialize_callback
    return pb.decode_delimited(proto.SimulationMessage(), packet)
  File "/Users/oliver/repos/phobos/scripts/phobos/pb.py", line 45, in decode_delimited
    length, position = _DecodeVarint32(data, 0)
  File "/Users/oliver/miniconda3/envs/dev/lib/python3.5/site-packages/protobuf-3.3.0-py3.5-macosx-10.7-x86_64.egg/google/protobuf/internal/decoder.py", line 121, in DecodeVarint
    b = six.indexbytes(buffer, pos)
IndexError: index out of range
```

mickvangelderen, 2017-07-25:
Opened a PR https://gitlab.com/bikelab/phobos-visualizer/merge_requests/1. Please review.

mickvangelderen, 2017-07-25:
PR was merged.

closed 2017-07-25 by mickvangelderen