oliverlee/phobos #247: FPS drop when using oculus

mickvangelderen, 2017-06-29:
Terrain is the problem, not sure how to fix it though.

mickvangelderen, 2017-06-29:
Added a toggle button to disable foliage rendering. Hope that fixes it. Need to test.

oliverlee, 2017-06-30:
It helps but it's still lower than 75. Goes from high 40s to low 50s.

oliverlee, 2017-06-30:
Using a build and choosing 'Good' for the graphics quality or lower results in a steady 75 FPS with and without foliage. 'Beautiful' and 'Fantastic' both result in FPS drop.

oliverlee, 2017-06-30:
FYI, the environment is drawn twice, once for each Oculus screen, which is probably resulting in the large FPS drop at the highest graphics quality settings.

mickvangelderen, 2017-06-30:
We are rendering next to nothing, it doesn't make sense. Even when rendering twice. I'm curious how many FPS we can get when uncapped.

oliverlee, 2017-07-06:
Need to reduce LOD of trees and foliage.

oliverlee, 2017-11-09:
This is resolved by replacing the Nvidia Quadro K2200 with a Nvidia GeForce GTX 1080 Ti. The Oculus DK2 has also been replaced with an Oculus CV1. Running the environment showswe are rendering consistently at 90 FPS. GPU temperature is around 60 C and GPU load is around 30-35%.

closed 2017-11-09 by oliverlee